sparkfun/SparkFun_MAX1704x_Fuel_Gauge_Arduino_Library #1: Max17048 with esp32 -wrong reading

Rj426, 2021-06-30:
Hi, I am trying to use Max17048 with esp32 board with the example provided here. 

i2c pins used -26,27

I am always getting 3.31 v on the serial monitor, although voltmeter is showing 4.14v.
`Voltage:3.31V Percentage: 0.04% Change Rate: 0.00%/hr Alert: 1 Voltage High Alert: 0 Voltage Low Alert: 1 Empty Alert: 0 SOC 1% Change Alert: 0 Hibernating: 1`

can you help me to provide some pointers.
Thanks

PaulZC, 2021-06-30:
Hi @Rj426 ,
Can you please tell us more about the hardware you are using:
Which ESP32 board? Which MAX17048 board? Which LiPo battery are you using? How do you have everything connected?
Thanks!
Paul

Rj426, 2021-06-30:
It’s esp32 wroom-32E board connected with max17048G. I am using Lipo 3.7v single cell for my connection. 
Connection is as below 
[image]

SDA and SCL  connection is going to esp32 pins 26 & 27.

PaulZC, 2021-06-30:
I think R19 and the way you have connected VDD to 3.3V are probably the cause of your problems. That looks more like the circuit for the MAX17049? Please check the MAX17048 datasheet for more details.

[image]

I'm closing this as this seems more likely to be a hardware issue, not an issue with the library.
Best wishes,
Paul

closed 2021-06-30 by PaulZC