sparkfun/SparkFun_MAX1704x_Fuel_Gauge_Arduino_Library #2: No need to wait after Wire.requestFrom().

Koepel, 2021-07-24:
In the file "SparkFun_MAX1704x_Fuel_Gauge_Arduino_Library.cpp" there is waiting with a timeout after the Wire.requestFrom(). That while-statement with the delay may be removed. There is no such thing as waiting after a Wire.requestFrom().

Explanation: [Common-mistakes#1](https://github.com/Koepel/How-to-use-the-Arduino-Wire-library/wiki/Common-mistakes#1)

PaulZC, 2021-07-26:
Thank you @Koepel ,
Changed in [this commit](https://github.com/sparkfun/SparkFun_MAX1704x_Fuel_Gauge_Arduino_Library/commit/6e2f3c7865bf3a0a5b4c9b86be5a3dcdcea3702b). I will include the changes in the next release after testing. Hopefully we won't see another case of [this issue](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/issues/3).
Best wishes,
Paul

closed 2021-07-26 by PaulZC

PaulZC, 2021-12-27:
Apologies for the delay! Resolved in v1.0.2...