sparkfun/SparkFun_MAX1704x_Fuel_Gauge_Arduino_Library #3: 0v reports 256% charge

dotMorten, 2021-09-19:
Was reading the voltage and percentage, and one reading was completely off, leading to this result:
```c
  voltage = lipo.getVoltage();
  soc = lipo.getSOC();
  Serial.println("Battery Voltage = " + String(voltage) + " (" + String(soc) + "%)");
```
output:
```
08:16:44.717 -> Battery Voltage = 4.17 (97.98%)
08:16:45.739 -> Battery Voltage = 4.17 (97.98%)
08:16:46.711 -> Battery Voltage = 4.17 (97.98%)
08:16:48.706 -> Battery Voltage = 0.00 (256.00%)
08:16:49.727 -> Battery Voltage = 4.17 (97.98%)
08:16:50.748 -> Battery Voltage = 4.17 (97.98%)
```

Looks like there's a bug where 0v equates to 256% instead of 0%.

PaulZC, 2021-09-20:
Hi Morten (@dotMorten ),

Thanks for reporting this.

Looking at the ```getSOC``` code:

```
  uint16_t soc;
  float percent;
  soc = read16(MAX17043_SOC);
  percent = (float)((soc & 0xFF00) >> 8);
  percent += ((float)(soc & 0x00FF)) / 256.0;
```

The only way percent can be 256% is if soc is 0xFFFF. Which is clearly invalid... And also very weird since ```read16``` will return 0x000, not 0xFFFF, if the I2C transmission fails.

```getVoltage``` uses a completely separate I2C transmission ( ```vCell = read16(MAX17043_VCELL);``` ).

Best guess, the MAX1704n was having a momentary bad day, or there was a drop in I2C communication?

I can 'fix' this by checking for soc == 0xFFFF and then returning zero %. But maybe it is better to leave the code as-is? 256% is more obviously wrong. The battery could legitimately report zero % when depleted.

Does this happen often? It would be really useful to be able to see the debug messages around this. Could you repeat with ```lipo.enableDebugging();``` enabled?

Very best wishes,
Paul

dotMorten, 2021-09-20:
Thanks for this analysis. Yesterday I also observed a momentary 5.12v which also resulted in 256%.

PaulZC, 2022-12-15:
Hi @dotMorten ,

I’m closing this… The 256% issue with the ESP was due to a lack of pull-up resistors on the I2C bus. With pull-ups, the issue goes away.

Hope you’re doing ok.
Very best wishes,
Paul

closed 2022-12-15 by PaulZC