sparkfun/SparkFun_MAX1704x_Fuel_Gauge_Arduino_Library #4: isConnected always returns false

kiowadriver, 2021-12-26:
The function isConnected checks the version of the MAX1704x to see if it responds.  I believe the logic has an error.  The getVersion() function returns a 0x3 (binary 110)when successful.  However, the current logic masks off the result with a binary "1000" which always returns false. 

 //Get version should return 0x001_
    //Not a great test but something
    //Supported on 43/44/48/49
    if (getVersion() & (1 << 4))
      return true;
  }

Instead should the logic just simply be looking for a 3?
 //Get version should return 0x001_
    //Not a great test but something
    //Supported on 43/44/48/49
    if (getVersion() == 0x3)
      return true;
  }

PaulZC, 2021-12-26:
Hi @kiowadriver ,

Thank you for reporting this. It looks like ```isConnected()``` was written for the MAX17048/9:

[image]

If ```getVersion()``` is returning 0x3, I guess you must be using a MAX17043/4?

Please confirm which chip you are using - thank you.

Best wishes,
Paul

kiowadriver, 2021-12-26:
> <sparkfun/SparkFun_MAX1704x_Fuel_Gauge_Arduino_Library/issues/4/1001150734
> @github.com>

Correct.

I am using 17043

PaulZC, 2021-12-27:
Hi @kiowadriver ,
Thanks again for raising this issue. It is resolved in v1.0.2.
Best wishes,
Paul

closed 2021-12-27 by PaulZC