sparkfun/SparkFun_MAX1704x_Fuel_Gauge_Arduino_Library #6: Difficulty getting ALRT interrupt to fire.

jumper-d-1981, 2022-01-26:
Using this library even though there are a couple different ones available, this one looks to be most functional.

USE CASE:
I have this thing reporting the voltage and SOC for my battery perfectly every time.  That is not the problem, the problem is that I cannot get the ALRT pin to work.  I have it connected to pin IO18/pin30.  I do not see where it tells me where to actually plug it in, but this is the pin we have it on.  The question is, I thought it was suppose to change state when it drops below a certain level and send an alert.  I can not find example code to use this function.  Could someone hook me up with some simple code to flash a light or send a serial print that the voltage is below a certain amount?  OR even just give me code to get this device to send the alert at all?  I am using example 3 and have tried the others...none will send alert.

Thanks.

PaulZC, 2022-01-26:
Hi @jumper-d-1981 ,

Please tell us which MAX1704x board you are using and which microcontroller board. Please also tell us how you have them connected. Are you using Qwiic plus a separate jumper wire for the ALRT?

Best wishes,
Paul

jumper-d-1981, 2022-01-26:
I am using max17048 with ESP32S from AITHINKER.  Using the Stemma connector i2c defaults...i.e. slc/sda(IO22 and 21) and the interrupt/ALRT pin is set to be IO18.

After some time, and some digging I realized that I think I was doing something incorrectly.

Using the example 4 "kitchen sink" I was able to see that there was another way to pull an Alert.  So I used the Alert lipo.isVoltageLow.  Put that into a variable and created an if statement.  It seems to work now.  I was having difficulty because the datasheet was not clear on how to get that alert to fire.

Now that it is firing when voltage is under 3.6, all we need to do is adjust threshold and we should be good.

I appreciate your time, but I think I got it.

Thanks.

PaulZC, 2022-01-26:
I'm glad that's working for you.

You can clear the individual alert flags in the CONFIG and STATUS registers by calling the ```get``` or ```is``` function with the clear parameter set to true. E.g. ```lipo.isVoltageLow(true)``` will return true if the STATUS VL flag is set, and will clear it within the same call.

Best wishes,
Paul

closed 2022-01-26 by PaulZC