sparkfun/SparkFun_MAX1704x_Fuel_Gauge_Arduino_Library #14: getSOC() returns value greater than 100.0f

Jakub-Bielawski, 2023-02-13:
This function should return values 0 - 100% but sometimes for MAX170438 it returns >100.0

PaulZC, 2023-02-13:
Hi Jakub (@Jakub-Bielawski ),

Please check if you are seeing this issue: #3 

Thank you,
Paul

Jakub-Bielawski, 2023-02-21:
Hi, I saw this but there is a difference between this two solutions. In my case MAX17048 returns soc >100% in consecutive samples when the battery is charging by external device. I know it's wired but I can share you my settings of MAX17048 if you want.

Thanks for your reply

PaulZC, 2023-02-22:
Hi Jakub (@Jakub-Bielawski ),

Do you only see SOC > 100% when the battery is charging? Perhaps only when the battery is charging and close to fully-charged? When charging is complete, does the SOC stabilize at close to 100%?

I don't think this is anything to worry about. For example, I see SOC reported as 104% when I have no battery connected:

[image]

Maybe try a different battery? ;-)

I hope this helps,
Paul

Jakub-Bielawski, 2023-02-22:
Hi (@PaulZC), 
I see SOC > 100% when battery is charging, after disconnect it goes below 100%. 
I deal with it by simply check if it's greater and assign 100%. 
I just want to report it to you to check it.

And by the way, thanks for your replay!
Jakub

PaulZC, 2023-02-22:
Hi Jakub,

The SOC comes straight from the MAX17048. There is nothing in our library which could alter the value. I think the 17048 is reporting >100% because the voltage during charging can be slightly higher than the voltage from a fully charged battery not connected to a charger.

I am going to close this issue because there is nothing we can do to correct it. But I will pin it to make it easier for other users to find. Just in case anyone notices the same issue.

Very best wishes,
Paul

closed 2023-02-22 by PaulZC

Jakub-Bielawski, 2023-02-22:
Hi Paul,
thanks for your answer!

Best wishes,
Jakub