sparkfun/SparkFun_MAX1704x_Fuel_Gauge_Arduino_Library #15: Issue with Connection

dominickairmyne, 2024-01-26:
Any idea why I would constantly read 5v and 256% charge? I have been troubleshooting for hours no idea what to do!

PaulZC, 2024-01-26:
Hi @dominickairmyne ,

What hardware are you using? Is this a SparkFun board - or a custom board? How are VDD and CELL connected?

Please check the [datasheeet](https://cdn.sparkfun.com/assets/5/2/7/6/6/MAX17048-MAX17049.pdf). Are you maybe using a MAX17048 to read a 2-cell battery?

Best,
Paul

dominickairmyne, 2024-01-26:
Hi! Thanks for the response! I am using the sparkfun board. VDD is connected to the 3v3 pin of my esp32 and CELL is connected to the Battery + terminal. The board uses the MAX17043 and it is a 1 cell 3.7V 1000mah Lipo. Again thank you so much and let me know if there is any other info you need!

PaulZC, 2024-01-26:
I think the VDD connection is causing your issue. We connect both VDD and CELL to battery +ve:

[image]

[image]

Closing... Please re-open if you need more help with this.

Best wishes,
Paul

closed 2024-01-26 by PaulZC

dominickairmyne, 2024-01-29:
Thanks for the advice! Just a quick question, Should I be concerned about overvolting my ADC in this case? What level are the I2C pins pulled up too? I remember seeing a warning on the Spark fun setup page about this. Sorry if this is not the right place to ask!

PaulZC, 2024-01-29:
I think your "5V" is a red herring due to CELL being higher than VDD. Connect both pins to battery +ve and you should start seeing sensible voltages etc..
The I2C bus is pulled up to 3.3V via the external pull-up resistors (2.2K in the above diagram). The SCL and SDA chip pins themselves are open-collector and can only pull-down. Check that you have the pull-ups somewhere, either close to the MAX17043 or they could be on your ESP32 board. I2C won't work without them.

dominickairmyne, 2024-01-29:
Okay thanks again got it! I really appreciate the help! So just to confirm this should be my hardware hookup?

[image]

PaulZC, 2024-01-29:
Looks good to me,
Best wishes,
Paul

dominickairmyne, 2024-02-11:
Hi back again! I did this setup above and am running the basic example code from this library, but this is what I keep getting in serial: 

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4
MAX17043 Example
SFE_MAX1704X::isConnected: retrying...
SFE_MAX1704X::isConnected: retrying...
SFE_MAX1704X::isConnected: retrying...
SFE_MAX1704X::isConnected: failed to detect IC!
begin: isConnected returned false
MAX17043 not detected. Please check wiring. Freezing. 

--------------------------------------------------------------

My GPIO's are differnt than default for I2C So I put Wire.begin(46,3); for my pins instead of Wire.begin();. Neither work. 

To verify I also wrote a I2C scanner script which found the device at 0x32, so it seems it might be a library issue?

Any ideas? Sorry I'm having so much trouble getting it to work!

PaulZC, 2024-02-11:
Hi @dominickairmyne ,

Please check you are using the correct GPIO numbers, not the physical package pin numbers. Both 46 and 3 look suspicious, but I don't know which exact ESP32 board you are using.

Your code needs to look something like this:

```
// Global

#include <Wire.h>
int mySCL = 15; // Use this GPIO pin for SCL
int mySDA = 12; // Use this GPIO pin for SDA
TwoWire myI2C = TwoWire(1); // Use I2C 1

#include <SparkFun_MAX1704x_Fuel_Gauge_Arduino_Library.h>
SFE_MAX1704X myLipo(MAX1704X_MAX17043); // Create a MAX17043

// setup

myI2C.begin(mySDA, mySCL);
myLipo.begin(myI2C);
```

I hope this helps,
Paul

dominickairmyne, 2024-02-25:
Thanks for the advice! I am using the ESP32-S3 Wroom Cam Board. I'll attach a pinout to this comment. I tried the default GPIO's and many others using code that integrates your setup code in the comment above but I still end up returning 0V, 0% charge. Any clear issues with my approach or any idea of what to change? Sorry for so many questions I'm really having bad luck with this haha.  Here is my code: 
------------------------------------------------------------------------------------------------------------------------------------------

#include <Wire.h>
int mySCL = 9; // Use this GPIO pin for SCL
int mySDA = 8; // Use this GPIO pin for SDA
TwoWire myI2C = TwoWire(1); // Use I2C 1

#include <SparkFun_MAX1704x_Fuel_Gauge_Arduino_Library.h>
SFE_MAX1704X myLipo(MAX1704X_MAX17043); // Create a MAX17043

double voltage = 0; // Variable to keep track of LiPo voltage
double soc = 0; // Variable to keep track of LiPo state-of-charge (SOC)

// setup

void setup() {

  Serial.begin(115200); // Start serial, to output debug data
  while (!Serial)
    ; //Wait for user to open terminal
  Serial.println(F("MAX17043 Example"));

  myI2C.begin(mySDA, mySCL);
  myLipo.begin(myI2C);
}

void loop() {
  
  voltage = myLipo.getVoltage();
  soc = myLipo.getSOC();

  // Print the variables:
  Serial.print("Voltage: ");
  Serial.print(voltage);  // Print the battery voltage
  Serial.println(" V");

  Serial.print("Percentage: ");
  Serial.print(soc); // Print the battery state of charge
  Serial.println(" %");

  delay(500);

}
-----------------------------------------------------------------------------------------------------------------------------------------
Also sorry if this is a obvious mistake but can't I assign any free GPIO's to do I2C? What is the issue of declaring using differnt pins if they are unused? 

[image]

PaulZC, 2024-02-25:
Hi @dominickairmyne ,

My code snippet was too simplistic. You aren't checking if the ```begin``` is successful. So you don't really know if you are connected to the fuel gauge or not. Please have a look at [the examples](https://github.com/sparkfun/SparkFun_MAX1704x_Fuel_Gauge_Arduino_Library/blob/6ea1765a513dde65e433dde12a70fced73b65fed/examples/Example1_Simple/Example1_Simple.ino#L46-L54). The debug messages will give you more help.

You should be able to use any _free_ pin. Do you have the camera connected? If you do, you can't use any of the CAM pins. You also can't use the PSRAM pins. Or the USB pins. For that board, 46 and 3 are marked as strapping pins - best avoided too. Maybe try 47 and 21?

Check you have the correct board selected in the Arduino IDE. I don't know if Freenove have their own board package? If they do, use the correct board from there. If not, make sure you have ESP32S3 Dev Module selected.

I hope this helps,
Paul